Hellod035/LeggedLab #3: 训练开始一直报资源加载问题，请问有办法解决吗？

rocketman123456, 2025-03-12:
Error Info
```
2025-03-12 07:19:18 [10,377ms] [Warning] [omni.kvdb.plugin] Disabling key-value database because another kit process is locking it
2025-03-12 07:19:18 [10,681ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 7374781440
2025-03-12 07:19:18 [10,681ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid false, within: false
2025-03-12 07:19:18 [10,681ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : decrement: 167690, decrement size: 7301033856
2025-03-12 07:19:18 [10,681ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : New limit 9748724 (slope: 439, intercept: 13179904)
2025-03-12 07:19:18 [10,681ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 4287352704
2025-03-12 07:19:18 [10,681ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid true, within: true
2025-03-12 07:19:18 [10,777ms] [Error] [omni.kit.renderer.plugin] IRendererImpl::updateReferenceFromTextureData getResourcesDescriptorSet failed for pool 1
[10.845s] app ready
2025-03-12 07:19:18 [11,047ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults-transient/meshlights/forceDisable'
[16.552s] Simulation App Startup Complete
Setting seed: 42
[16.760s] [ext: omni.physx.fabric-106.5.7] startup
[INFO] Generating terrains randomly took : 1.442283 seconds
2025-03-12 07:23:32 [265,270ms] [Warning] [rtx.neuraylib.plugin] [EntityResolver] FileDatasourceInputStream failed to open: http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/IsaacLab/Materials/TilesMarbleSpiderWhiteBrickBondHoned/TilesMarbleSpiderWhiteBrickBondHoned.mdl
2025-03-12 07:23:32 [265,271ms] [Error] [rtx.neuraylib.plugin] [MDLC:COMPILER]   1.0   MDLC   comp error: http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/IsaacLab/Materials/TilesMarbleSpiderWhiteBrickBondHoned/TilesMarbleSpiderWhiteBrickBondHoned.mdl(1,1): C100 "mdl" expected
2025-03-12 07:23:32 [265,271ms] [Warning] [rtx.neuraylib.plugin] Loading MdlModule to DB (OmniUsdMdl) failed: ::Z6Ahttp_3A_2F::Z58omniverse_2Dcontent_2Dproduction_2Es3_2Dus_2Dwest_2D2_2Eamazonaws_2Ecom::Assets::Isaac::ZD1_34_2E5::Isaac::IsaacLab::Materials::TilesMarbleSpiderWhiteBrickBondHoned::TilesMarbleSpiderWhiteBrickBondHoned
2025-03-12 07:23:32 [265,271ms] [Warning] [rtx.neuraylib.plugin] Loading transaction committed (this thread). MdlModule is NOT in the DB (OmniUsdMdl): ::Z6Ahttp_3A_2F::Z58omniverse_2Dcontent_2Dproduction_2Es3_2Dus_2Dwest_2D2_2Eamazonaws_2Ecom::Assets::Isaac::ZD1_34_2E5::Isaac::IsaacLab::Materials::TilesMarbleSpiderWhiteBrickBondHoned::TilesMarbleSpiderWhiteBrickBondHoned
2025-03-12 07:23:32 [265,272ms] [Error] [omni.usd] USD_MDL: in LoadModule at line 242 of ../../source/plugins/omni.usdMdl/neuray.cpp -- 'rtx::neuraylib::MdlModuleId' for 'http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/IsaacLab/Materials/TilesMarbleSpiderWhiteBrickBondHoned/TilesMarbleSpiderWhiteBrickBondHoned.mdl' is Invalid

2025-03-12 07:23:32 [265,272ms] [Error] [omni.usd] USD_MDL: in GetSdrFromDiscoveryResult at line 183 of ../../source/plugins/omni.usdMdl/moduleRegistry.cpp -- Module: 'http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/IsaacLab/Materials/TilesMarbleSpiderWhiteBrickBondHoned/TilesMarbleSpiderWhiteBrickBondHoned.mdl' with version '1' not found in 'MdlModuleRegistry::ModuleDataMap'.

2025-03-12 07:29:02 [594,856ms] [Warning] [rtx.neuraylib.plugin] [EntityResolver] FileDatasourceInputStream failed to open: http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/IsaacLab/Materials/TilesMarbleSpiderWhiteBrickBondHoned/TilesMarbleSpiderWhiteBrickBondHoned.mdl
2025-03-12 07:29:02 [594,856ms] [Error] [rtx.neuraylib.plugin] [MDLC:COMPILER]   1.0   MDLC   comp error: http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/IsaacLab/Materials/TilesMarbleSpiderWhiteBrickBondHoned/TilesMarbleSpiderWhiteBrickBondHoned.mdl(1,1): C100 "mdl" expected
2025-03-12 07:29:02 [594,857ms] [Warning] [rtx.neuraylib.plugin] Loading MdlModule to DB (rtx_scope0) failed: ::Z6Ahttp_3A_2F::Z58omniverse_2Dcontent_2Dproduction_2Es3_2Dus_2Dwest_2D2_2Eamazonaws_2Ecom::Assets::Isaac::ZD1_34_2E5::Isaac::IsaacLab::Materials::TilesMarbleSpiderWhiteBrickBondHoned::TilesMarbleSpiderWhiteBrickBondHoned
2025-03-12 07:29:02 [594,857ms] [Warning] [rtx.neuraylib.plugin] Loading transaction committed (this thread). MdlModule is NOT in the DB (rtx_scope0): ::Z6Ahttp_3A_2F::Z58omniverse_2Dcontent_2Dproduction_2Es3_2Dus_2Dwest_2D2_2Eamazonaws_2Ecom::Assets::Isaac::ZD1_34_2E5::Isaac::IsaacLab::Materials::TilesMarbleSpiderWhiteBrickBondHoned::TilesMarbleSpiderWhiteBrickBondHoned
2025-03-12 07:29:02 [594,857ms] [Error] [omni.hydra] Failed to create MDL shade node for prim '/World/ground/terrain/visualMaterial/Shader'. createMdlModule failed.
```

Hellod035, 2025-03-12:
This is because Isaaclab will download resources from Amazon by default, which is sometimes not feasible when the network is restricted. The solution is:

1. Download and install Omniverse Launcher from https://install.launcher.omniverse.nvidia.com/installers/omniverse-launcher-linux.AppImage

2. Install Nucleus in Omniverse Launcher

3. Modify the default asset directory in IsaacLab/source/isaaclab/isaaclab/utils/assets.py

chage
`NUCLEUS_ASSET_ROOT_DIR = carb.settings.get_settings().get("/persistent/isaac/asset_root/cloud")`
to
`NUCLEUS_ASSET_ROOT_DIR = "omniverse://localhost/NVIDIA/Assets/Isaac/4.5"`

rocketman123456, 2025-03-13:
Thanks a lot, that really solve the problem

closed 2025-03-13 by rocketman123456