Hellod035/LeggedLab #4: cannot run tensorboard to checkout training data

rocketman123456, 2025-03-12:
running `tensorboard --logdir logs/` gives:
```
TensorFlow installation not found - running with reduced feature set.
Traceback (most recent call last):
  File "/home/rocketsky/.local/lib/python3.10/site-packages/pandas/__init__.py", line 26, in <module>
    from pandas.compat import (
  File "/home/rocketsky/.local/lib/python3.10/site-packages/pandas/compat/__init__.py", line 26, in <module>
    from pandas.compat.numpy import is_numpy_dev
  File "/home/rocketsky/.local/lib/python3.10/site-packages/pandas/compat/numpy/__init__.py", line 21, in <module>
    raise ImportError(
ImportError: this version of pandas is incompatible with numpy < 1.22.4
your numpy version is 1.21.5.
Please upgrade numpy to >= 1.22.4 to use this pandas version

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/rocketsky/.local/bin/tensorboard", line 8, in <module>
    sys.exit(run_main())
  File "/home/rocketsky/.local/lib/python3.10/site-packages/tensorboard/main.py", line 39, in run_main
    tensorboard = program.TensorBoard(plugins=default.get_plugins())
  File "/home/rocketsky/.local/lib/python3.10/site-packages/tensorboard/default.py", line 86, in get_plugins
    return get_static_plugins() + get_dynamic_plugins()
  File "/home/rocketsky/.local/lib/python3.10/site-packages/tensorboard/default.py", line 121, in get_dynamic_plugins
    return [
  File "/home/rocketsky/.local/lib/python3.10/site-packages/tensorboard/default.py", line 122, in <listcomp>
    entry_point.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2471, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/rocketsky/.local/lib/python3.10/site-packages/torch_tb_profiler/plugin.py", line 21, in <module>
    from .profiler import RunLoader
  File "/home/rocketsky/.local/lib/python3.10/site-packages/torch_tb_profiler/profiler/__init__.py", line 5, in <module>
    from .loader import RunLoader
  File "/home/rocketsky/.local/lib/python3.10/site-packages/torch_tb_profiler/profiler/loader.py", line 14, in <module>
    from .data import DistributedRunProfileData, RunProfileData
  File "/home/rocketsky/.local/lib/python3.10/site-packages/torch_tb_profiler/profiler/data.py", line 18, in <module>
    from .kernel_parser import KernelParser
  File "/home/rocketsky/.local/lib/python3.10/site-packages/torch_tb_profiler/profiler/kernel_parser.py", line 7, in <module>
    import pandas as pd
  File "/home/rocketsky/.local/lib/python3.10/site-packages/pandas/__init__.py", line 31, in <module>
    raise ImportError(
ImportError: C extension: None not built. If you want to import pandas from the source directory, you may need to run 'python setup.py build_ext' to build the C extensions first.
```

Hellod035, 2025-03-12:
This seems to be a problem with the installation of tensorboard. The tensorboard version installed by default in Isaac Lab is tested to be feasible. Please create a new conda environment and reinstall IsaacLab and Legged Lab.

rocketman123456, 2025-03-13:
i solve it by remove system custom installed packages, located in ~/.local, it conflit with conda env

closed 2025-03-13 by rocketman123456