Hellod035/LeggedLab #7: walk issue

Xyunze, 2025-03-14:
hello I appreciate your fantastic work. I tried to train H1 with your code with rsl_rl, isaaclab and Isaac Sim 4.5.  Have you ever get some error about Calling getByPassRenderSkelMeshProcceing for prim/Visuals/Command/velocity_current.proto0_Cylinder_id0 that not been populated. and the trained policy can't let the robot move. Thanks

Hellod035, 2025-03-14:
It looks like this is some warning from the command debug arrow in the render, which should not affect the policy. If it fails to render, set commands.debug_vis to False.

For the case where the robot does not move, we trained h1_flat for 50,000 iterations in the default settings, and the robot can follow the velocity command normally. Please increase the number of training iterations in your settings. If you still have problems after 50,000 iterations, please feel free to report it.

Xyunze, 2025-03-14:
Thank for your repl

Xyunze, 2025-03-14:
Thank for your quick reply, I trained with num_envs=10000 and max_iterations=3000. I will train another 3000 iterations and find out. The warning of command doesn't go away even if I turn the debug_vis = false on the base_config.py

Hellod035, 2025-03-14:
Thank you for your feedback. If the warning does not affect the use, you can ignore it. Normally, you should be able to see the arrow above the robot's head.
During training, num_envs=4096 may be a more common practice. We look forward to your feedback after increasing the number of training iterations. In the middle of our training process, there are also cases where the robot can stand but cannot walk. It is recommended that you remain patient. If the tracking_lin_vel reward is above 0.7, it means that it is working normally.

Xyunze, 2025-03-18:
My problem is fixed. I just need more training iterations. Thank for your help.

closed 2025-03-18 by Xyunze