Hellod035/LeggedLab #9: legged_lab develop issue

vamWu, 2025-03-24:
Thank you for your open-source contributions.

First, I trained a model using your repository at [https://github.com/Hellod035/legged_lab](https://github.com/Hellod035/legged_lab) with the following command:

```
python legged_lab/scripts/train.py --task=g1_flat --headless --logger=tensorboard --num_envs=4096 --num_steps_per_env 24 --num_mini_batces 5
```

I executed 30,000 iterations and then manually paused the training because I observed that the mean_reward had converged, stabilizing around 25.6 in TensorBoard. Subsequently, I used play.py:

```
python legged_lab/scripts/play.py --task=g1_flat
```

This process generated the corresponding `policy_me.pt` file.

I then deployed the model on physical hardware using the repository [https://github.com/Hellod035/unitree_deploy_py](https://github.com/Hellod035/unitree_deploy_py) along with the official Unitree tutorials. When I used the `policy.pt` file from your repository, everything worked as expected – the G1 robot was able to be debugged, walk, and stand normally. However, when I replaced the `policy.pt` file with `policy_me.pt`, the following error occurred:

1. Could you please explain the reason for this issue and provide any reference solutions?
2. Is it possible to manually stop the training process once convergence is observed, rather than waiting until the maximum number of iterations is reached?

I am new to Embodied AI, so my questions may be somewhat basic. Thank you for your time and assistance.

---

Below is the error traceback I encountered:

```
Traceback of TorchScript, original code (most recent call last):
 File "/media/newLinDisk/Reproduce/isaaclab/IsaacLab/source/isaaclab_rl/isaaclab_rl/rsl_rl/exporter.py", line 76, in forward
   def forward(self, x):
       return self.actor(self.normalizer(x))
              ~~~~~~~~~~ <--- HERE
 File "/home/xt/anaconda3/envs/isaaclab/lib/python3.10/site-packages/torch/nn/modules/container.py", line 250, in forward
   def forward(self, input):
       for module in self:
           input = module(input)
                   ~~~~~~ <--- HERE
       return input
 File "/home/xt/anaconda3/envs/isaaclab/lib/python3.10/site-packages/torch/nn/modules/linear.py", line 125, in forward
   def forward(self, input: Tensor) -> Tensor:
       return F.linear(input, self.weight, self.bias)
              ~~~~~~~~ <--- HERE
RuntimeError: mat1 and mat2 shapes cannot be multiplied (1x96 and 960x512)
```

Hellod035, 2025-03-24:
Hi, it looks like you are doing the right thing. The only small problem is that num_mini_batces is usually set to 4, but you set it to 5. This will not matter if the robot can move normally in simulation after training. 
As for deployment,the error is caused by different observation lengths. In the delopy code, I used a history length of 1 step by default in the delopy code, and a history length of 10 steps by default in the training code. You can modify
[unitree_deploy_py/configs/g1.yaml](https://github.com/Hellod035/unitree_deploy_py/blob/main/configs/g1.yaml)

`history_length: 1 `
to
`history_length: 10`

I believe it will fit perfectly into the training code (although I haven't tested it yet). Please give us your feedback in time, I hope this helps you.

vamWu, 2025-03-25:
Thank you for your response.

1. After carefully reviewing the code, I also noticed the issue during training. For the G1_flat task, the `actor_obs_history_length` is set to 10 by default (inherited from the `BaseEnvCfg` class), whereas for the G1_rough task, the `actor_obs_history_length` is set to 1 (overridden in the `G1FlatEnvCfg`).

2. I would like to ask about the rationale behind setting `actor_obs_history_length` to 10 or 1 during training. What considerations were taken into account when choosing these values, and how can we evaluate which setting would yield better performance when deploying on real hardware?

Hellod035, 2025-03-25:
For complex terrain tasks, long-term memory is said to bring better results, so the network uses LSTM and one-step history is used accordingly. [RSS2021 Blind Stair](https://roboticsconference.org/2021/program/papers/061/index.html)
As a result, according to human preferences, if you think it is better, then it is actually better.

closed 2025-03-25 by vamWu