Hellod035/LeggedLab #10: Question about playing trained policy

MARMOTatZJU, 2025-03-27:
Thanks for open-sourcing this project which is truely a missing piece of the robotics community?

I have successfully trained a `h1_flat` policy with the instruction in README. Howver, I was not able to find any clue about playing the trained policy. I read the `play.py` script and tried the following command which corresponded to the best of my understanding:

```python
python legged_lab/scripts/play.py --task=h1_flat --num_env 1
```

However, I encountered errors related to numpy as follows:

```text
AttributeError: module 'numpy.linalg._umath_linalg' has no attribute '_ilp64'
```

Seeing that I am new to IsaacSim/IsaacLab, I could not figure it by myself. I would appreciate if you could shed more light on ways to play trained policy and put it in the README.

Thanks in advance!

Hellod035, 2025-03-27:
Hi, If you have trained h1_flat, you can view the policy with
`python legged_lab/scripts/play.py --task=h1_flat`
which will load the latest trained policy by default.
You seem to have an export problem, which is not common. Please make sure you have installed isaaclab and leggedlab from your newly created conda/mamba environment according to the
[isaaclab tutorial](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/index.html) and [leggedlab tutorial](https://github.com/Hellod035/legged_lab).
 Please test this again in new environment:
`python legged_lab/scripts/train.py --task=h1_flat --headless --logger=tensorboard --num_envs=64
`and
`python legged_lab/scripts/play.py --task=h1_flat
`If the problem persists, please contact me as soon as possible and I will work with you to further troubleshoot the problem.

YuHoChau, 2025-03-28:
This problem is from the latest version of rsl_rl, try re-installing this version: [https://github.com/leggedrobotics/rsl_rl/tree/v2.2.4](https://github.com/leggedrobotics/rsl_rl/tree/v2.2.4)

MARMOTatZJU, 2025-03-28:
@YuHoChau Thanks for your advice! Originally, I also guessed that it could be a problem caused by changing interface.

I installed Isaac Sim and Isaac Lab from pip (https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/isaaclab_pip_installation.html) which is an easy and clean way. I just checked my environemtn and strangely found that `rsl_rl` can be imported by my side but no version is tagged, neither by running `python -c "import rsl_lib;print(rsl_lib.__version)"` or `pip list|grep rsl`. I will tell the `rsl_rl` version once I find it.

@Hellod035 Could you please provide a recommended installation method of Isaac-Sim and Isaac-Lab since currently multiple official methods exsist for installing them? I think the problem root might lie in the Python package versions.

MARMOTatZJU, 2025-03-28:
@MARMOTatZJU I checked with command from you by my side. My rsl_rl version is 2.3.0 and it was installed along with pip installation of Isaac Lab v2.0.2.

Hellod035, 2025-03-28:
Thank you for your question. If you have any further questions, please inform us as soon as possible.

gyt290811, 2025-03-31:
Hello! I understand you're saying you're encountering the same problem in the "play" phase after training your Isaac Lab environment, and you're wondering what could be causing this.

[image]

MARMOTatZJU, 2025-03-31:
@gyt290811  Not the same problem in your screenshot. However, your issue seems to be related to depndencies' versions (e.g. IsaacSim, IsaacLab, etc.). Maybe re-check versions of your runtime environment?

MARMOTatZJU, 2025-03-31:
@Hellod035 I fetched the latest commit and re-run `play.py` today. The latest script can be run without error. Thank you for your update!

gyt290811, 2025-03-31:
@MARMOTatZJU @Hellod035 I'm not sure whether this is a version issue？

Hellod035, 2025-03-31:
@gyt290811 The problem should come from the resource loading issue from amazon, please follow [this issue](https://github.com/Hellod035/LeggedLab/issues/3) to set the local asset catalog and try again, I believe the problem will be solved.

gyt290811, 2025-03-31:
> [@gyt290811](https://github.com/gyt290811) The problem should come from the resource loading issue from amazon, please follow [this issue](https://github.com/Hellod035/LeggedLab/issues/3) to set the local asset catalog and try again, I believe the problem will be solved.

@Hellod035 Yeah, the '_problem should come from the resource_' issue is resolved, but there was another problem even before that, as you can see in the image, and I'm completely stuck on how to solve it!
[image]

gyt290811, 2025-04-01:
> [@gyt290811](https://github.com/gyt290811) The problem should come from the resource loading issue from amazon, please follow [this issue](https://github.com/Hellod035/LeggedLab/issues/3) to set the local asset catalog and try again, I believe the problem will be solved.

The solution to [this issue](https://github.com/Hellod035/LeggedLab/issues/3)  has three parts/points. Are these steps to be done in sequence, or are they three options to choose from?

Hellod035, 2025-04-01:
@gyt290811 Please follow 1.2.3. in order. The prerequisite for each step is that the previous step has been completed.

gyt290811, 2025-04-01:
> [@gyt290811](https://github.com/gyt290811) Please follow 1.2.3. in order. The prerequisite for each step is that the previous step has been completed.

For Nucleus installation in Omniverse Launcher: Do I search for "NUCLEUS" in the `Exchange `tab (like for "Omniverse Nucleus Wrapp" and "Omniverse Nucleus Navigator"), or is it installed from a dedicated "`NUCLEUS`" area? Are there any installation guides I can refer to? (New Issac Lab user here, previously worked with gym).

gyt290811, 2025-04-01:
> > [@gyt290811](https://github.com/gyt290811) Please follow 1.2.3. in order. The prerequisite for each step is that the previous step has been completed.
> 
> Thanks to the solution in the issue, I installed it and now the results are playing as expected/normally. Thank you very much!
> 
> If I chaged `NUCLEUS_ASSET_ROOT_DIR = carb.settings.get_settings().get("/persistent/isaac/asset_root/cloud`) to `NUCLEUS_ASSET_ROOT_DIR = "omniverse://localhost/NVIDIA/Assets/Isaac/4.5`, will it have consequences for the training process afterwards? Because the results from the previous training weren't good, and I don't know why, I want to train it again.
It just fall down with the trained policy

[image]

Hellod035, 2025-04-01:
@gyt290811 Please make sure you did not use `--num_envs=64` when training and retest the training. The correct training command is `python legged_lab/scripts/train.py --task=h1_flat --logger=tensorboard`, which will use 4096 environments. It is expected to get good results after 10,000 iterations.

MARMOTatZJU, 2025-04-03:
@Hellod035 Seeing that my question has been answered, you may consider closing this issue.