Hellod035/LeggedLab #11: Extremely Long Waiting Time When Training Template in Isaac Lab

Jack-Zhang1101, 2025-03-29:
warning like this：
2025-03-29 09:26:02 [2,003ms] [Warning] [omni.kvdb.plugin] Disabling key-value database because another kit process is locking it
Setting seed: 42
[INFO] Generating terrains randomly took : 0.771058 seconds
2025-03-29 09:30:10 [250,022ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /builds/omniverse/usd-ci/USD/pxr/usd/usd/stage.cpp -- In </Visuals/Command/velocity_goal/arrow>: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/Props/UIElements/arrow_x.usd@  for reference introduced by @anon:0x4561bda0:World0.usd@</Visuals/Command/velocity_goal/arrow>. (recomposing stage on stage @anon:0x4561bda0:World0.usd@ <0x45620fe0>)

2025-03-29 09:30:10 [250,050ms] [Warning] [omni.usd] Warning: in GetNodeDiscoveryResults at line 136 of /builds/omniverse/usd-ci/USD/pxr/usd/usdShade/shaderDefUtils.cpp -- Unable to resolve info:sourceAsset </UsdPreviewSurface.info:mdl:sourceAsset> with value @UsdPreviewSurface.mdl@.

This is my first time using Isaac Lab. When I try to train the template,“python legged_lab/scripts/train.py --task=h1_flat --headless --logger=tensorboard --num_envs=64” it takes about ten minutes before it starts. Could anyone please tell me the reason? Thank you so much for your help!

Hellod035, 2025-03-29:
Please check the commands in the [IsaacLab installation guide](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/index.html) to ensure that IsaacLab is installed correctly.
It is normal for IsaacLab to take a long time to start for the first time. If the problem of taking a long time to open persists, and this is a phenomenon unique to training in LeggedLab and not in IsaacLab, please communicate with me in time.

Jack-Zhang1101, 2025-03-29:
> Please check the commands in the [IsaacLab installation guide](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/index.html) to ensure that IsaacLab is installed correctly. It is [...]

ok,thanks,I will have a try.

Jack-Zhang1101, 2025-03-31:
> ok,thanks,I will have a try.

I have fix the isaacsim bugs by download Local Assets Packs following:https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_faq.html# , thanks a lot for you kind help.

closed 2025-04-01 by Hellod035