Hellod035/LeggedLab #12: how to make mixed control type for differwnt joint？

rocketman123456, 2025-03-30:
Hi，this repository is really helpful for me to start transfer from isaacgym. but i cannot find a correct way to control my wheel legged robot with position control for leg joint and velocity control for wheel. how to do this with this framework？

Hellod035, 2025-03-31:
```
self.robot.set_joint_position_target(processed_actions)
self.robot.set_joint_velocity_target(processed_velocity)
self.robot.set_joint_effort_target(processed_effort)
```
As far as I know, you can use these three commands in implicit joints. Their calculation method is PD plus feedforward torque control. I hope it will be helpful to you.

rocketman123456, 2025-03-31:
it is useful，i make a wrong assumption that usd joint order is same as urdf joint order，so i cannot control the selected joints at first

closed 2025-04-01 by rocketman123456