Hellod035/LeggedLab #15: Cannot import name 'RslRlRndCfg' from 'isaaclab_rl.rsl_rl'

mggeng, 2025-04-10:
```
Traceback (most recent call last):
  File "/workspace/isaaclab_extension_template/legged_lab/scripts/train.py", line 27, in <module>
    from legged_lab.envs import *  # noqa:F401, F403
  File "/workspace/isaaclab_extension_template/legged_lab/envs/__init__.py", line 1, in <module>
    from legged_lab.envs.base.base_env import BaseEnv
  File "/workspace/isaaclab_extension_template/legged_lab/envs/base/base_env.py", line 1, in <module>
    from legged_lab.envs.base.base_env_config import BaseEnvCfg
  File "/workspace/isaaclab_extension_template/legged_lab/envs/base/base_env_config.py", line 9, in <module>
    from isaaclab_rl.rsl_rl import RslRlOnPolicyRunnerCfg, RslRlPpoActorCriticCfg, RslRlPpoAlgorithmCfg, RslRlRndCfg, RslRlSymmetryCfg  # noqa:F401
ImportError: cannot import name 'RslRlRndCfg' from 'isaaclab_rl.rsl_rl' (/workspace/isaaclab/source/isaaclab_rl/isaaclab_rl/rsl_rl/__init__.py)
```

I build the docker issac_lab_template and bind the respository source dirctory to /workspace/issaclab_externsion_tempplate/(refer to https://github.com/isaac-sim/IsaacLabExtensionTemplate). The problem occurred running:
```
python legged_lab/scripts/train.py --task=h1_flat --headless --logger=tensorboard --num_envs=64
```
i think this is because the moudle RslRlRndCfg cause the problem, but `RslRlRndCfg` only start to work since issaclab_rl.rsl_rl main branch. what is the best way to solve it?
Best Regards!

Hellod035, 2025-04-11:
I'm sorry you're having this problem. I'm not familiar with docker, but I'll provide some insights to help you debug.
Legged Lab is actually built on top of the latest version of IsaacLab, and it has nothing to do with IsaacLab Extension Template. In theory, you only need to compile a docker image with IsaacLab. Then install Legged Lab using a python interpreter with IsaacLab installed.
It looks like you're having an IsaacLab version issue. Maybe you should git pull in the container or use the latest IsaacLab version when compiling the image? Instead of specifying a version number, just clone the latest version, for example, `git clone https://github.com/isaac-sim/IsaacLab.git`.
I hope my insights can help you

mggeng, 2025-04-11:
Thanks a lot!
I have tried install IsaacLab from isaacsim image before, but it failed due to Internet problem.
I successfully run the respository training with conda on host environent. I will try to tring in docker environment for coperation development.
If there are any updates, I'll keep it posted in this issue thread.
Best Regards!

Hellod035, 2025-04-11:
So glad this was helpful to you 😀.

closed 2025-04-12 by mggeng