Hellod035/LeggedLab #17: 机器人推理阶段不运动

bingqi6, 2025-04-15:
训练3万次之后，推理阶段可视化，机器人只是站在哪里，却不运动

Hellod035, 2025-04-15:
This is not common, and it is likely that one of your modifications caused the problem. If you trained on the Legged Lab code, please tell me your training and play commands and I will troubleshoot with you.

bingqi6, 2025-04-15:
I didn't modify any parameters and used g1 inside the file directly as my robot for training, he doesn't move. After that I copied other people's code for training, the same code can walk after training on other people's computers but not on mine!

mggeng, 2025-04-15:
I think I encountered a similar issue before, and the solution was to adjust the weightings for x/y-axis followin (may be as well as z). I hope this can help. Maybe .. But this is weird while you append the explaination and i start to confuse.

bingqi6, 2025-04-15:
One more question, I run the example code . /isaaclab.sh -p scripts/demos/h1_locomotion.py and it walks stably without any problems I don't think it should be my environment!

closed 2025-04-19 by Hellod035