Hellod035/LeggedLab #18: Asymmetrical foot movement after training.

mggeng, 2025-04-15:
Hello, I am here again.

I trained my modole (12 Dofs half Humaniod)，Here is my envs.
```
    track_lin_vel_xy_exp = RewTerm(func=mdp.track_lin_vel_xy_yaw_frame_exp, weight=2.2, params={"std": 0.5})
    track_ang_vel_z_exp = RewTerm(func=mdp.track_ang_vel_z_world_exp, weight=1.0, params={"std": 0.5})
    lin_vel_z_l2 = RewTerm(func=mdp.lin_vel_z_l2, weight=-1.0)
    ang_vel_xy_l2 = RewTerm(func=mdp.ang_vel_xy_l2, weight=-0.5)
    energy = RewTerm(func=mdp.energy, weight=-1e-3)
    dof_acc_l2 = RewTerm(func=mdp.joint_acc_l2, weight=-2.5e-7)
    action_rate_l2 = RewTerm(func=mdp.action_rate_l2, weight=-0.01)
    undesired_contacts = RewTerm(func=mdp.undesired_contacts, weight=-1.0, params={"sensor_cfg": SceneEntityCfg("contact_sensor", body_names="(?!.*_roll_and_FOOT.*).*"), "threshold": 1.0})
    fly = RewTerm(func=mdp.fly, weight=-1.0, params={"sensor_cfg": SceneEntityCfg("contact_sensor", body_names=".*_roll_and_FOOT.*"), "threshold": 1.0})
    body_orientation_l2 = RewTerm(func=mdp.body_orientation_l2, params={"asset_cfg": SceneEntityCfg("robot", body_names=".*base_link.*")}, weight=-2.0)
    flat_orientation_l2 = RewTerm(func=mdp.flat_orientation_l2, weight=-1.0)
    termination_penalty = RewTerm(func=mdp.is_terminated, weight=-200.0)
    feet_air_time = RewTerm(func=mdp.feet_air_time_positive_biped, weight=1.5, params={"sensor_cfg": SceneEntityCfg("contact_sensor", body_names=".*_roll_and_FOOT.*"), "threshold": 0.4})
    feet_slide = RewTerm(func=mdp.feet_slide, weight=-0.5, params={"sensor_cfg": SceneEntityCfg("contact_sensor", body_names=".*_roll_and_FOOT.*"), "asset_cfg": SceneEntityCfg("robot", body_names=".*_roll_and_FOOT.*")})
    feet_force = RewTerm(func=mdp.body_force, weight=-3e-3, params={"sensor_cfg": SceneEntityCfg("contact_sensor", body_names=".*_roll_and_FOOT.*"), "threshold": 500, "max_reward": 400})
    # feet_too_near = RewTerm(func=mdp.feet_too_near_humanoid, weight=-2.0, params={"asset_cfg": SceneEntityCfg("robot", body_names=[".*_roll_and_FOOT.*"]), "threshold": 0.2})
    feet_stumble = RewTerm(func=mdp.feet_stumble, weight=-2.0, params={"sensor_cfg": SceneEntityCfg("contact_sensor", body_names=[".*_roll_and_FOOT.*"])})
    dof_pos_limits = RewTerm(func=mdp.joint_pos_limits, weight=-2.0)
    joint_deviation_hip = RewTerm(func=mdp.joint_deviation_l1, weight=-0.15, params={"asset_cfg": SceneEntityCfg("robot", joint_names=[".*_joint_roll.*", ".*_joint_yaw.*"])})
    # joint_deviation_arms = RewTerm(func=mdp.joint_deviation_l1, weight=-0.2, params={"asset_cfg": SceneEntityCfg("robot", joint_names=[".*waist.*", ".*_shoulder_roll.*", ".*_shoulder_yaw.*", ".*_wrist.*"])})
    joint_deviation_legs = RewTerm(func=mdp.joint_deviation_l1, weight=-0.02, params={"asset_cfg": SceneEntityCfg("robot", joint_names=[".*_joint_pitch.*", ".*_joint_knee.*",".*_joint_ankle.*"])})
    lfeet_orientation = RewTerm(func=mdp.feet_orientation, weight=-0.2, params={"asset_cfg": SceneEntityCfg("robot", body_names=[".*l_roll_and_FOOT.*"])})
    rfeet_orientation = RewTerm(func=mdp.feet_orientation, weight=-0.2, params={"asset_cfg": SceneEntityCfg("robot", body_names=[".*r_roll_and_FOOT.*"])})
    feet_keep_distance = RewTerm(func=mdp.feet_keep_distance, weight=-0.8, params={"asset_cfg": SceneEntityCfg("robot", body_names=[".*_roll_and_FOOT.*"])})
```
Here is the rewards added. (From torobo humaniod with little change)
```
def feet_orientation(env: BaseEnv, asset_cfg: SceneEntityCfg = SceneEntityCfg("robot")) -> torch.Tensor:
    """ Penalize the difference between the orientation of the feet and the root """
    asset = env.scene[asset_cfg.name]
    body_quat = asset.data.body_quat_w[:, asset_cfg.body_ids, :].reshape(-1, 4)
    root_quat = asset.data.root_quat_w
    angle_diff = torch.abs(math_utils.quat_error_magnitude(body_quat, root_quat))
    return angle_diff

def feet_keep_distance(env: BaseEnv, asset_cfg: SceneEntityCfg = SceneEntityCfg("robot")) -> torch.Tensor:
    """ Penalize if the distance between the two feet about the y-axis is not in the range of 0.22m to 0.32m """
    asset = env.scene[asset_cfg.name]
    body_pos = asset.data.body_pos_w[:, asset_cfg.body_ids, :]
    root_pos = asset.data.root_pos_w[:, :]
    root_quat = math_utils.yaw_quat(asset.data.root_quat_w)
    root_to_body_1 = body_pos[:, 0, :].squeeze(1) - root_pos[:, :]
    root_to_body_2 = body_pos[:, 1, :].squeeze(1) - root_pos[:, :]
    root_to_body_1_b = math_utils.quat_rotate_inverse(root_quat, root_to_body_1)
    root_to_body_2_b = math_utils.quat_rotate_inverse(root_quat, root_to_body_2)
    feet_y_distance = torch.abs(root_to_body_1_b[:, 1] - root_to_body_2_b[:, 1])
    rew = torch.where((feet_y_distance > 0.08 * 2) & (feet_y_distance < 0.16 * 2), 0, 1)
    return rew
```

While i played the policy. The right foot shows noticeable lifting action while the left foot doesn't. I haven't observed any visible asymmetries in either the code or the model.
Unlike other training policy, the gaint phase is not considered. I think this is maybe the reason. 

By the way, are you going to develop sim2sim like mujoco in the project?

Best Rerads!

Guang

Hellod035, 2025-04-15:
Hi, mujoco support is not planned, I think physX is good enough. To train your own humanoid, my debugging advice is to start with the smallest reward, which is the reward that leggedlab already has, and then consider additional rewards when the effect is not satisfactory. I hope this can help you debug

mggeng, 2025-04-15:
Thanks for your help, any progress will be posted here.

closed 2025-05-15 by mggeng