Hellod035/LeggedLab #19: g1_flat training problem

Luo624, 2025-04-21:
When I reproduced the g1_flat code, it appeared that there was a 0.9 angular velocity tracking reward in the 1000 iterations, but the linear velocity tracking reward kept oscillating between 0.3 and 0.5. The linear velocity tracking reward did not rise to 0.9 until 14,000 iterations. When I play the trained model, the robot's leg-lifting height is very low. I estimate it is no more than 0.03 meters. How can I solve this problem?

Hellod035, 2025-04-22:
Hi, this is normal. Due to the fixed seed (42), the training results are completely reproducible, which is expected. The height of the foot will not affect the deployment performance.

Luo624, 2025-04-22:
Do you mean that even if the trained model has a very low leg-lifting height in the simulation, it will have a higher leg-lifting height in the g1 robot in real world?

Hellod035, 2025-04-22:
The foot lift height will not be higher when deployed, but the current foot lift height is acceptable to me.

Luo624, 2025-04-22:
I understand what you mean. Thank you for your reply

closed 2025-04-22 by Hellod035