Hellod035/LeggedLab #25: 最新更新main的版本用不了

JakeFishcode, 2025-06-01:
还想请教一下，height_scan是否还需要进一步蒸馏？

Hellod035, 2025-06-02:
Please describe the problem in detail, otherwise the author will solve the problem specifically.

closed 2025-06-02 by Hellod035

JakeFishcode, 2025-06-02:
```
Traceback (most recent call last):
  File "/home/jk/X2/LeggedLab/legged_lab/scripts/train.py", line 101, in <module>
    train()
  File "/home/jk/X2/LeggedLab/legged_lab/scripts/train.py", line 97, in train
    runner.learn(num_learning_iterations=agent_cfg.max_iterations, init_at_random_ep_len=True)
  File "/home/jk/anaconda3/envs/isaaclab/lib/python3.10/site-packages/rsl_rl/runners/on_policy_runner.py", line 204, in learn
    obs, rewards, dones, infos = self.env.step(actions.to(self.env.device))
  File "/home/jk/X2/LeggedLab/legged_lab/envs/base/base_env.py", line 240, in step
    reward_buf = self.reward_manager.compute(self.step_dt)
  File "/home/jk/Lab/IsaacLab/source/isaaclab/isaaclab/managers/reward_manager.py", line 149, in compute
    value = term_cfg.func(self._env, **term_cfg.params) * term_cfg.weight * dt
  File "/home/jk/X2/LeggedLab/legged_lab/mdp/rewards.py", line 30, in track_lin_vel_xy_yaw_frame_exp
    vel_yaw = math_utils.quat_apply_inverse(
AttributeError: module 'isaaclab.utils.math' has no attribute 'quat_apply_inverse'. Did you mean: 'quat_rotate_inverse'?
2025-06-02 05:08:39 [8,086ms] [Warning] [omni.fabric.plugin] gFabricState->gUsdStageToSimStageWithHistoryMap had 1 outstanding SimStageWithHistory(s) at shutdown
2025-06-02 05:08:39 [8,105ms] [Warning] [carb] Recursive unloadAllPlugins() detected!

```